sameezch/OMNI-BAND #5: Print to serial from arduino through a Processing User Interface.

gianlazz, 2017-05-28:


gianlazz, 2017-05-28:
@sameezch Go ahead and share your findings here in the comment section. Links, and example code will be great or just personal notes/comments.

sameezch, 2017-05-29:
https://learn.sparkfun.com/tutorials/connecting-arduino-to-processing/to-processing

gianlazz, 2017-05-29:
@sameezch We're almost done with this one. You pretty much finished it and soon we should be able to merge this in somewhere and document the merge on this issue as we close it.

gianlazz, 2017-05-29:
@sameezch there's an id for each commit, merge, etc that you can copy and paste into here and it will act as a hyperlink to the commit or merge it'self.

gianlazz, 2017-05-29:
@sameezch so when we complete a designated feature within a branch we comment the link to when we do a pull request(or merge?) into the master or other more root branch and close the issue.

gianlazz, 2017-05-31:
4a5a6563724460839b5abf58739948cb122dd9f1

closed 2017-05-31 by gianlazz