sameezch/OMNI-BAND #10: Adjust the dropdown width

sameezch, 2017-06-03:
Control P5 is being stupid. It's not letting me adjust the width of the drop down menu.

sameezch, 2017-06-04:
c7a06bd222ce2dfbe0adff2d2e0208deed7f6337

Look here for the code

gianlazz, 2017-06-04:
http://www.sojamo.de/libraries/controlP5/examples/controllers/ControlP5dropdownList/ControlP5dropdownList.pde

gianlazz, 2017-06-04:
@sameezch 

Solved with adding the line below to the void customize dropdownlist method:
`ddl.setWidth(200);`

In this commit:
2e04ae9a5cb47f94be00a6711d1c76a834ccd93b

If there are future issues with dropdownList width sizing related the length of the port name we may have to reopen this issue and switch from a hard coded width to using another one of the functions listed above.

closed 2017-06-04 by gianlazz