sameezch/OMNI-BAND #11: Heart rate sensor as ControlP5 line chart

gianlazz, 2017-06-03:
http://www.sojamo.de/libraries/controlP5/examples/controllers/ControlP5chart/ControlP5chart.pde

accepts float variable

sameezch, 2017-06-04:
ee27e02728fdb9c266c8bbf8ac9dbac70f99689a

I have added the chart to the processing UI Code for the heart rate. It's just a line graph

gianlazz, 2017-06-04:
Same as the issue mentioned in this link:
https://github.com/sameezch/OMNI-BAND/issues/12

gianlazz, 2017-06-04:
Take Heart Rate serial integer value 0-1000 and turn less than 550 into 0.0 and over 550 into 1.0 (as float variables) so that we can show the high points on the sinewave as heartbeats and low points as pauses.

gianlazz, 2017-06-04:
How do I get the serial heart rate int of 0-1024 into the processing sketch?

Solved in Omni.pde in this commit: 4a5a6563724460839b5abf58739948cb122dd9f1

gianlazz, 2017-06-04:
Making new issue to merge in the Omni.pde code to display arduino serial text in a processing window.

gianlazz, 2017-06-04:
"Created Implement-HeartRate-Float-Into-Graph branch for myself to work on this feature in."

resuming work here as a duplicate issue was started for this feature https://github.com/sameezch/OMNI-BAND/issues/16

gianlazz, 2017-06-04:
How do I pipe in the serial output data from just one sensor? Do I have to parse the text for a keyword?

gianlazz, 2017-06-04:
Below is the line of code where I'll implement my heartrate float.

 myChart.setData("incoming", new float[100]);

gianlazz, 2017-06-04:
@sameezch Here's some sudo code and leads on search terms that may lead to supporting discussions or documentation:

sameezch, 2017-06-04:
https://processing.org/reference/libraries/io/I2C.html

sameezch, 2017-06-04:
Let's try i2c.list() and see what it prints. From there we can specify sensors to different variables. As for the HR we can use the serial function.

gianlazz, 2017-06-04:
I like that!

sameezch, 2017-06-05:
Hey! We should use port.bufferUntil(). 

https://processing.org/reference/libraries/serial/Serial_bufferUntil_.html

We will just set different characters for each sensor

- [ ] Create float variables

- [ ] Set different characters for each sensor in the Arduino code and upload it

- [ ] Parse the string as a float